mistergreen/WAAC #5: Need some help?

LuckyLuzz, 2018-02-25:
Hey,
I very much like you software, it's unique. I'd like to contribute by creating a shield for the ESP32 ESP32-DEVKITC. But before I start I'd like to talk about the pinout and the functions. 

I'd say:
  1: IN PH-Sensor
  2: IN EC-Sensor
  3: IN 3x Temp Sensor DS18B20 
  4: OUT DAC1 0-10V
  5: OUT DAC2 0-10V
  6: OUT PWM Dosing Pumpe 1
  7: OUT PWM Dosing Pumpe 2
  8: OUT PWM Dosing Pumpe 3
  9: OUT PWM LED Channel 1
10: OUT PWM LED Channel 2
11: OUT PWM LED Channel 3
12: OUT PWM LED Channel 4

Plus internal SD Card pins and I²C for IO expensions. Also I'd like to implement a 20x4 LCD Display with Rotary encoder for basic informations (IP Adresse, Temp, PH, EC....)

I chose that pinout because it is the most basic need for every aquarium, everything else can be implementet using I²C and IO expension chips like MCP23017 or PCA9685.

What do you think?

mistergreen, 2018-02-25:
Thanks for the interest. I'm also working on a library for the DHT22/11 (humidity/temperature sensor) for terrarium users too or you can use it to detect water leaks in closed spaces. A pH sensor would be good. A shield with screw connectors would be great for secure connections. Right now I'm testing with simple jumper wires but it could come loose in practical applications.

LuckyLuzz, 2018-02-25:
A display is mandatory for me, the LCD itslef would not take up extra pins because it would be connected over the reserved I²C, but the rotary encoder would take 3 pins. Will try to get the encoder running over a MCP23107.

mistergreen, 2018-02-25:
You can have code cycle through the LCD info. That would eliminate the need for the rotary encoder. The MCP23107 can only do digital IO? no analog or PWM?

LuckyLuzz, 2018-02-25:
Yes, it only can IO, no PWM. 

What is your SD Card setup?

mistergreen, 2018-02-25:
I'm using SPI for SD. pin 5/SS, pin 18/SCk, pin 23/MOSI, pin 19/MISO, and 3v3 & ground.

LuckyLuzz, 2018-02-25:
[image]
[ESP32_WAAC_Pinout.xlsx](https://github.com/mistergreen/WAAC/files/1756690/ESP32_WAAC_Pinout.xlsx)

If this is alright with you, I am going to start with the board layout.

mistergreen, 2018-02-25:
looks good.

LuckyLuzz, 2018-02-26:
The PH Sensor is going to be that: https://github.com/SparkysWidgets/MinipHHW  

Is there a other way we could communicate?

mistergreen, 2018-02-26:
Most I2C devices have 2 or more usable address.
I did a quick look and the MinipH uses 0x4D as the default which is fine and won't conflict with the PCA9685.

mistergreen, 2018-02-26:
Another I2C pH sensor is https://www.sparkfun.com/products/10972
It's pricey but scientific grade.

LuckyLuzz, 2018-02-27:
yeah I know that product, but the price is just to high (30€ Probe + 5€ Chip for 150€) and also where is the fun in buying it? I think it`s much more fun to take a open source version and make something out of it or at least proof or decline the concept.

Very good to know that there will be no adress conflict. Have you seen that I added 433Mhz RF to the Pinout? Should be awesome to switch wireless socket with it and as feature there is no need to wire 220V (110V)  relais (good for people with no technical skills). [https://goo.gl/rvqZqe]

mistergreen, 2018-02-27:
Nice. More wireless the better.
So any pH probe will work with the minPH?

LuckyLuzz, 2018-02-27:
I mean I can't guarantee that till I tried it, but they should, because they working more or less with the same basic principle. https://en.wikipedia.org/wiki/PH_meter

mistergreen, 2018-03-28:
I've added support for MCP23107 IO in the core code. I'm not a hardware guy so when I daisy chain the PCA9685 to the MCP23107, it wouldn't work. By themselves it's fine. I2C devices are so finicky. I have to be careful which V+ & Ground bus I use or partly works.

LuckyLuzz, 2018-10-01:
Hey, the hardware problems could be a wrong value of the pullup resistors. With value did you use? 

Then I am glad to say that the first hardware prototype in manufactuing. The whole thing is my bachelor thesis, after testing the hardware I have to start coding. So I'd like to ask you if you would mind helping me out with informations to the html stuff. Which IDE did you use, what software did you use in general to create the website?

As the work is finished it is going to be published on github too. I updated the pinout in the upper post.

Regards
Thomas

LuckyLuzz, 2018-10-01:
thanks, the whole thing has to be ready in febuary. The I²C pull up depends on the cable capacity and the bus speed, something between 2k2 and 10k should work on a normal breadboard. It's a bit try an error. 

I am gone try Google Web Designer and Wordpress to see what works best for me. And for the rest of the code you have used Arduino IDE?

mistergreen, 2018-10-01:
yes, I used the Arduino IDE.

LuckyLuzz, 2018-10-01:
Thank you very much.

mistergreen, 2018-10-01:
Oh, also.. If and when you need to edit/create the C, C++ libraries, I use Visual Studio.

LuckyLuzz, 2018-10-15:
Hey, is there a reason why you choose to use jQuery Mobile as UI Framework? It hasn't been updated for years, so I thought it might be good to switch to a different Framework.
I have read a lot about different frameworks but cannot dicide which one I wane use...

AngularJS
Bootstrap
Material components
???

What would be your dicesion if you would start the project from scratch?

mistergreen, 2018-10-15:
Yeah, it was made years ago so I used jQuery mobile to have the 'app' feel. The I used jQuery as the web interface so Bootstrap would make a good replacement. It could save you time instead of starting from scratch.