mistergreen/WAAC #6: SPIFFS instead of SD Card

LuckyLuzz, 2018-03-04:
Hey,
I was searching for a way to dump the sd card from the web server and found out that it is possible to save files in the 4MB Flash of the ESP32. So what do you think about that?

http://www.lucadentella.it/en/2017/09/30/esp32-22-spiffs/

mistergreen, 2018-03-04:
Yeah, I saw that a while back. I'm still in development so this process is cumbersome in the work flow. It's easier to change files on an SD card. I also want people to mod their own images and css easily.

Maybe if people want to go this direction, they can but I think I'll keep the SD.

LuckyLuzz, 2018-03-17:
allright

LuckyLuzz, 2018-11-16:
Hey, with Platformio and VS Code it's accaly easier to load files in spiffs and use it instead of pluging the sd card in and out the whole time as you develope. There is just one limitation, die data Folder must not have a size bigger than 1MB. So in order to put all html and js files info Spiffs if there are bigger than 1MB they need to be packed in gzip.

mistergreen, 2018-11-16:
Ahh, cool. I was playing with platform and Vscode. It's very nice to work with with minor edits to 'arduino' code.