mistergreen/WAAC #9: ESP32 Login doesn't work

LuckyLuzz, 2018-05-24:
Hi,
the login window is not working, nothing happend after i type admin and waac.

Seriel says:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
Connecting to SAKE
........Connected to wifi
IP Address: 10.1.176.22
SUCCESS - Found index.htm file.
waiting for sync
Transmit NTP Request
Receive NTP Response
memory 118852
file not found
//sha1.js
file not found
//sha1.js

mistergreen, 2018-05-24:
I added a new login.htm
Try that. It was a password issue.

mistergreen, 2018-05-25:
I checked it on my ESP32. It works now.

closed 2018-05-25 by mistergreen