mistergreen/WAAC #10: E-Mail test show wrong result

LuckyLuzz, 2018-05-27:
Hey,
the test mail function in "Settings" is somehow not working. The E-Mail is sending but it shows that it is not and then the website loading but nothing happend.

Could you check that?

Thanks :)

mistergreen, 2018-05-28:
Can you be more specific? 

> The E-Mail is sending but it shows that it is not and then the website loading but nothing happend.

I don't understand. I entered my email, hit test, I received the email in my inbox. No error messages popped up. I hit 'save' at the bottom of the settings page to saving the email.

LuckyLuzz, 2018-05-28:
Alright, what I did is:
e-mail --> test --> I get the mail --> waiting circle but nothing happend -->press "save" --> nothing happend 

[image]

IP Address: 10.1.176.22
SUCCESS - Found index.htm file.
waiting for sync
Transmit NTP Request
Receive NTP Response
memory 118852
user pass
14981
HTTP/1.1 200 OK
Server: nginx/1.12.2
Date: Mon, 28 May 2018 13:50:12 GMT
Content-Type: text/html
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding

d
php mail sent
0

then press "cancel" --> Error message

[image]
[image]

then the server is not reacheble anymore --> I have to press "reset" to start it again

If you like we could try a teamviewer session.

Regards

mistergreen, 2018-05-28:
Hey make sure you have the latest files from the master.
When I saw 

> d
> php mail sent
> 0

In your Serial output, it looks like from an older file.

LuckyLuzz, 2018-05-29:
Hey,
well I can save the adresse before hiting "test". But as I hit "test" it has the same result as before.

Newest Master is installed.

mistergreen, 2018-05-30:
Can you confirm that the ESP32 crashed? It should output in the serial monitor or forces you to log back into the website. 

It would be good if I have the same result as you so I can debug it.

LuckyLuzz, 2018-05-30:
Hey,
it definetly crashes. There is no other serial output as the one that i posted already. What do you mean with log back into the Website? It loads the site but nothing happend.

So how about teamviewer session? You can connect to my Computer and check things there. Just sent me a mail with a ID request.

mistergreen, 2018-05-30:
Go ahead and update to the latest master. I think you got caught in an infinite loop when getting feedback from my server. When my ESP32 crashes, it restarts. Let's see if this fixes it.

LuckyLuzz, 2018-06-02:
I did update to latest master. 
esp still crash, 
this time I didn't get any TEST E-Mail,
the serial output is changed:

SUCCESS - Found index.htm file.
waiting for sync
Transmit NTP Request
Receive NTP Response
memory 118848
user pass
27098
success ajax
HTTP/1.1 200 OK
Server: nginx/1.14.0
Date: Sat, 02 Jun 2018 08:16:51 GMT
Content-Type: text/html
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding

22
to: lucky.luzz@gmx.dephp mail sent
0

mistergreen, 2018-06-02:
The email sent to your inbox sometimes takes a while. It's just how the internet works :)
Do you have the latest ESP32 core libraries by the way?
https://github.com/espressif/arduino-esp32

I updated mine a few months ago.

LuckyLuzz, 2018-06-04:
yeah i got the mails, but still same problem. Also I found out that I have the same crash problem as I want to add a device. But that happend just sometimes. I have sent you a mail, please check you inbox.

I installed al the arduino stuff just last week, so its up-to-date.

LuckyLuzz, 2018-06-04:
hey, your bugfix worked, i didn't see that you have changed something within the wwwsettings lib so I didn't replaced it with master. Sorry...

I guess it can be closed.

mistergreen, 2018-06-04:
cool. good to know it works.

closed 2018-06-04 by mistergreen