mistergreen/WAAC #12: Problem with SDCard

dodolcool, 2018-11-10:
Hello
Im trying to make this project but after success upload the code and connect to network an error message always appear
ERROR - SD card initialization failed!
could you specify which sdcard module used on this project?  im using esp32 and catalex microsd module. with P23=MOSI P19=MISO P18=SCK and P8=CS.

Also can you running PWM without any external module? which pin can i used? this is the firs time i used esp32 and it's make me confused >.<

mistergreen, 2018-11-10:
Try pin 5 for SS/CS. 
Yes, you can use native PWM. That's any 'A' pin, (A0-A14).

dodolcool, 2018-11-10:
ups sorry my bad. I mean P5 not P8. for CS and the error occurs like i mention above.

mistergreen, 2018-11-10:
Try running a simple example SD card sketch and see if you have the same problem. 
Make sure the module is connected to 3v3 & ground. Check if the SD card is formatted with FAT32 and there are files on it.

asasi83, 2019-09-30:
I had the same issue, pins were connected like in [https://icircuit.net/arduino-interfacing-sd-card-esp-32/1857](url), and "SD card sketch" worked, but I still had "ERROR - SD card initialization failed!". The issue was solved when I added "FS.h".