mistergreen/WAAC #13: seems something is wrong with the esp32 code.

waltdavis, 2019-07-31:
This will not compile in Arduino IDE. i first tried on the current version and had many compile errors. So after reading more about it i saw your mention of needing version 1.8.1.  i uninstalled newer version and installed 1.8.1... still throws a lot of errors during compile. none of wich i can figure out as the code is pretty complex..any help would be greatly appreciated

createHallSensor' was not declared in this scope
'class OneWireSensor' has no member named 'getLowerTarget'
these are the first two errors..

mistergreen, 2019-08-01:
I just verified the code with Arduino 1.8.5
No errors for me.

In your preference, check Show verbose.
Copy and paste the error.

waltdavis, 2019-08-04:
Thanks for the help. I uninstalled 1.8.1 and installed 1.8.5.  checked show verbose as you suggested. ran the verify and this is the entire error message. if i did the error message wrong i apologize. this is the first time I've asked for help so I'm still learning.

In file included from C:\Users\wdavi\OneDrive\Documents\esp32 vivarium\ESP32\WAAC_ESP32\WAAC_ESP32.ino:40:0:

C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\OneWire/OneWire.h:254:2: warning: #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture." [-Wcpp]

 #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."

  ^

In file included from C:\Users\wdavi\OneDrive\Documents\esp32 vivarium\ESP32\WAAC_ESP32\WAAC_ESP32.ino:37:0:

C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\Input/Input.h:17:1: error: expected class-name before '{' token

 {

 ^

In file included from C:\Users\wdavi\OneDrive\Documents\esp32 vivarium\ESP32\WAAC_ESP32\WAAC_ESP32.ino:38:0:

C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\InputMCP/InputMCP.h:18:1: error: expected class-name before '{' token

 {

 ^

In file included from C:\Users\wdavi\OneDrive\Documents\esp32 vivarium\ESP32\WAAC_ESP32\WAAC_ESP32.ino:41:0:

C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\OneWireSensor/OneWireSensor.h:17:1: error: expected class-name before '{' token

 {

 ^

C:\Users\wdavi\OneDrive\Documents\esp32 vivarium\ESP32\WAAC_ESP32\Analog.ino: In function 'void analogAjaxOutput(WiFiClient, Device*)':

Analog:6: error: 'Analog' does not name a type

        int tempInt = static_cast<Analog*>(device)->getPin();

                                  ^

Analog:6: error: expected '>' before '*' token

        int tempInt = static_cast<Analog*>(device)->getPin();

                                        ^

Analog:6: error: expected '(' before '*' token

Analog:6: error: expected primary-expression before '>' token

        int tempInt = static_cast<Analog*>(device)->getPin();

                                         ^

Analog:6: error: expected ')' before ';' token

        int tempInt = static_cast<Analog*>(device)->getPin();

                                                            ^

Analog:10: error: 'Analog' does not name a type

        client.print(static_cast<Analog*>(device)->getMin());

                                 ^

Analog:10: error: expected '>' before '*' token

[... 1292 lines cut ...]
Using library SD at version 1.0.5 in folder: C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\SD 
Using library FS at version 1.0 in folder: C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\FS 
Using library Time at version 1.5 in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\Time 
Using library Wire at version 1.0.1 in folder: C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\Wire 
Using library Device in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\Device (legacy)
Using library Adafruit_PWM_Servo_Driver_Library at version 1.0.2 in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\Adafruit_PWM_Servo_Driver_Library 
Using library AdaFruitPWM8 in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\AdaFruitPWM8 (legacy)
Using library Alert in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\Alert (legacy)
Using library DeviceDelegate in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\DeviceDelegate (legacy)
Using library Input in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\Input (legacy)
Using library InputMCP in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\InputMCP (legacy)
Using library Adafruit_MCP23017_Arduino_Library at version 1.0.3 in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library 
Using library MCP_helper in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\MCP_helper (legacy)
Using library OneWire at version 2.3.3 in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\OneWire 
Using library OneWireSensor in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\OneWireSensor (legacy)
Using library PCA_helper in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\PCA_helper (legacy)
Using library PWM4 in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\PWM4 (legacy)
Using library rBase64 at version 1.0.0 in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\rBase64 
Using library Relay in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\Relay (legacy)
Using library RelayMCP in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\RelayMCP (legacy)
Using library RelayPCA in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\RelayPCA (legacy)
Using library Shunt in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\Shunt (legacy)
Using library Video in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\Video (legacy)
Using library WebParser in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\WebParser (legacy)
Using library WWWsettings in folder: C:\Users\wdavi\OneDrive\Documents\Arduino\libraries\WWWsettings (legacy)

mistergreen, 2019-08-04:
Did you put the library files in the libraries folder? Doesn't look  like the Arduino IDE see the library files.

waltdavis, 2019-08-05:
after reading the errors closer, it might be the link to one drive. i will disable one drive and see if that solves the issue and report back...Thanks again!

waltdavis, 2019-08-05:
so after disabling one drive, i get a new error. it might actually work sson, just need to figure this one out.

Arduino: 1.8.9 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\wdavi\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\wdavi\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\wdavi\OneDrive\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\wdavi\AppData\Local\Temp\arduino_build_473087 -warnings=default -build-cache C:\Users\wdavi\AppData\Local\Temp\arduino_cache_352888 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\wdavi\OneDrive\Documents\esp32 vivarium\WAAC-master\WAAC-master\ESP32\WAAC_ESP32\WAAC_ESP32.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\wdavi\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\wdavi\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\wdavi\OneDrive\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\wdavi\AppData\Local\Temp\arduino_build_473087 -warnings=default -build-cache C:\Users\wdavi\AppData\Local\Temp\arduino_cache_352888 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\wdavi\OneDrive\Documents\esp32 vivarium\WAAC-master\WAAC-master\ESP32\WAAC_ESP32\WAAC_ESP32.ino
Using board 'esp32' from platform in folder: C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Using core 'esp32' from platform in folder: C:\Users\wdavi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Detecting libraries used...
"C:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/config" "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_trace" "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_update" "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/asio" "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bootloader_support" "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bt" "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/coap" "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/console" "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/driver" "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp-tls" "-IC:\\Users\\wdavi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include [... cut ...]
Using library Analog in folder: C:\Program Files (x86)\Arduino\libraries\Analog (legacy)
Using library DeviceDelegate in folder: C:\Program Files (x86)\Arduino\libraries\DeviceDelegate (legacy)
Using library HallSensor in folder: C:\Program Files (x86)\Arduino\libraries\HallSensor (legacy)
Using library Input in folder: C:\Program Files (x86)\Arduino\libraries\Input (legacy)
Using library InputMCP in folder: C:\Program Files (x86)\Arduino\libraries\InputMCP (legacy)
exit status 1
Error compiling for board ESP32 Dev Module.

waltdavis, 2019-08-07:
so i have a partial success. i was able to install by converting to platformIO. after a bit of work that is. now when i run it, i can not connect to web page at ip address in code and I get this error in serial monitor.

......................Connected to wifi
IP Address: 192.168.1.177
SUCCESS - Found index.htm file.
waiting for sync
Transmit NTP Request
[E][WiFiUdp.cpp:170] beginPacket(): could not get host from dns: 0
No NTP Response
memory 243856
 I googled the error but was not able to find anything that made sense to me... btw when i get this to work, it will control 4 bio active tree frog tanks, one for a desert spiny lizard, and two crested gecko . I've made a nice board that contains a wroom-32d module with a mosfet and relay for 8 pins selected by a jumper. The boards works great if I run it on Blynk, but with limited function as i don't want to rely on that platform...Thanks again

mistergreen, 2019-08-08:
Glad to hear it's working
the error

> Transmit NTP Request
[E][WiFiUdp.cpp:170] beginPacket(): could not get host from dns: 0
No NTP Response

Is it can't connect to a Network Time Server... I think I used google but you can add another server.

I was playing with platformIO too. If you can post your changes for platformIO for others, that'll be great.

waltdavis, 2019-08-08:
I would be happy to post my changes.. would the NTP problem keep me from being able to access the web page to log in? I am really a NOOB with networking and am stumbling my through this. Hardware is my main ability. I added my ssid and passphrase, but left the code unaltered other than that. After upload and a reset of my board, I go to browser and enter 192.168.1.177. all i get is a "cant connect error" from browser. what am i missing?

mistergreen, 2019-08-09:
The NPT shouldn't interrupt the web server.

Can you open up an simple example web server script from Arduino? Enter in the necessary WIFI stuff and run that. See if you get the same connection error. Best to test out on a simple script.

waltdavis, 2019-08-09:
I have another board with a simple webserver script that I am using to control lights on a tank..it was running this morning. Although it looses connection often and requires a restart. I will run a simple sketch on the board I'm working on and see how that goes. I am waiting for more boards to get back from the board house, but if I need to I can use a dev. Board. If it will run a simple webserver any idea on where I could go next? 

After running this code through Platformio, I will never use arduino IDE again. Platform is much faster at uploads, and writing code seems easier...

mistergreen, 2019-08-09:
Import the simple web server to platformIO and run it from there too. I would say if your board can run a simple web server without connection loses, it can run WAAC. A good board shouldn't lose connections unless you turn it off.

waltdavis, 2019-08-10:
So i finally got it to run,,,the solution was simple, change the ip address, i used the one the the simple web server provided and change that in the WAAC code. ran first try! however the NTP time service is still down. ive tryed changing it via web interface, but still no luck...is there a better place to change the server in the code?

waltdavis, 2019-08-10:
it appears that im working on locan network with no access to internet for time server?

waltdavis, 2019-08-10:
also, is there a reason i have to set up all devices  each time i load the web page?

mistergreen, 2019-08-11:
> it appears that im working on locan network with no access to internet for time server?
Oh, can you surf the internet on your network? The Time Server is on the internet.

> also, is there a reason i have to set up all devices each time i load the web page?
The board must have reseted, lost power, or crashed. When that happens, all the settings in memory is lost.

waltdavis, 2019-08-11:
yes i can access the internet on my network, so the current settings are not stored? then if a reset is needed then you must set up the web page again?. would there be a good way to save the setting is spiffs if needed? i can work with it without saving, but cant use times without time server. im still a little lost in this code!

mistergreen, 2019-08-11:
I was going to make a way to say the settings to a text file and save it the SD card and restore on reset but it was too time consuming. I might revisit that if I have time. It's just tedious work.

On the time server, you should be able to change in the settings page but it you need to find the code for it it's in 
WAAC/ESP32/libraries/WWWsettings/WWWsettings.cpp
`line 23:     strcpy(NTPServer, "time.google.com"); //defaul NTP`

waltdavis, 2019-08-19:
so after playing around with this foe a week or so, its a pretty amazing piece of work. How much would it take to be able to save all settings from web page to spiffs? that way if a reset is needed or if you wanted to run your vivarium off line you could...i imagine that if a heater was on, and you lost network access for whatever reason, then the function to turn off heater at ? time or ? temp would not work. so over heating is a possibility. or imagine a mist pump...could flood a tank... or am i not seeing this right? so if the settings were stored in spiffs. then it would have all the info to keep running if a connection dropped...Again, I'm not a code guy, but I'm learning ...slowly through...

mistergreen, 2019-08-19:
Cool, glad it's working.. So you got the Network time to work?
The controller would still work with no network. The clocks and timers are still running. You just can't log into the webpage.It only loses the settings when you lose power. You can create a backup battery power for that.

I can save the settings to the SD card, or whatever (spiffs). It has to be formatted a specific way to work. 
When I have time I might work on that.

asasi83, 2019-09-30:
Hi, all wana say that this is a great project!!!  I just want to share some issues that I had when I was trying to compile this. I had the same issue as waltdavis...
in "Analog.cpp" and "HallSensor.cpp" there is "shunt.h" but it should be "Shunt.h" compiler output is not helpful at all :) 
Also if you get SD card error include "FS.h" I was always getting "ERROR - SD card initialization failed!"
And last one if you still cant compile switch to Linux :)