mistergreen/WAAC #14: Project build on Arduino IDE

Avilla83, 2020-01-09:
Hi,

I had to change the Sensor class name to SensorWaac in order to build the project. 
I am not sure which was the conflict with Sensor but I had the same issue on two PCs.
I already have the change in my fork (AVilla83) if needed to be integrated.

I would also like to extend this project with a fish feeded device.

mistergreen, 2020-01-09:
Thanks. That's interesting about the Sensor class.

Your fish feeding device should work fine with generic Relay device, digital on/off.

Avilla83, 2020-01-09:
Actually I need something more complex than the relay since I use a servo motor. I already have the code to control it but I need to integrate it. I will try as soon as I have some time. 

What I added now is a define to switch between SD card memory and SPIFFS memory. It is useful to use internal memory, especially during the development.

mistergreen, 2020-01-10:
Ah, see if the PWM class works for the servo.

Avilla83, 2020-01-13:
Thanks for your suggestions but I neeeded to add a new device (htm page and logic) that uses the servo library on my fork. 
Currently I did just few testing but it seems to send the right commands to the motor. I have not tested the scheduler yet.

Since the feeder should be resistant to possible power failures, what I would like to add is a non volatile storage for the configuration (general and devices). I am trying to use the ArduinoNvs library but I need some testing. 
Please let me know if you have any suggestion or if you already tried to store the WAAC configuration.

mistergreen, 2020-01-14:
I've looked at a JSON library, https://arduinojson.org/ to serialize settings to file. But I don't have anytime to mess with it. 
For now, you can give your mirco processor a backup battery power supply to save any power outage.

A PWM class should give you general control of a servo. It can't give the servo specific angles to move to.

Avilla83, 2020-01-26:
> I've looked at a JSON library, https://arduinojson.org/ to serialize settings to file. But I don't have anytime to mess with it.
> For now, you can give your mirco processor a backup battery power supply to save from any power outage.

I started to work on it on my fork, at the moment I can store part of the general configuration and the servo motor device. I plan to go ahead saving more configuration.
It is useful to set the ip and wifi settings in a json instead of in hte code.

> 
> A PWM class should give you general control of a servo. It can't give the servo specific angles to move to.

Since I use a modified Servo motor 9g I had to implement the new device.