mistergreen/WAAC #15: Persistent storage

Avilla83, 2020-02-14:
Hi,

I added storage capability to all sensors and devices. 
I know for sure that AdaFruitPWM8 is not stored completely and still needs some development.

Currently I tested only the ServoMotor class I added but it would be nice to have further tesing.
I am also checking how much memory is needed to serialize/deserialize the json for a correct memory allocation.

Would you like to collaborate or better integrate it in the main repository (maybe in a separate branch)?

I could start to send pull requests as soon as I implement something if you think it would be a good way to proceed.

mistergreen, 2020-02-14:
Did you use the json library for storage or some other method?
Yeah, I could start another branch to test thing.

Avilla83, 2020-02-14:
> Did you use the json library for storage or some other method?
Yes, I used ArduinoJson library to store.
I use a define to store internally to the ESP32 memory or externally to the SD card.

> Yeah, I could start another branch to test thing.
It would be helpfull especially for devices I don't have. I have changes for storage on a separate branch. I suspect it contains also the device I added to control the servo motor.
If you create the branch I send you the pull request.

mistergreen, 2020-02-14:
Ok, I've created a 'serialize device settings' branch.

urkrossi, 2020-06-03:
Hello I have great interest in their aquarium control. 
But I have to say that I am not a programmer.

I have tested the controller on a "Node MCU32" and "ESP32 Wrover" board, unfortunately I have problems saving the configuration on the SPIFFS or SD card memory.
When I add devices and save a OneWire thermometer, for example, it only works until the system is restarted. The configuration is not loaded after the restart.

When I write the devices.json in the following format to memory.
`>`
[
{
  "numDevices": 1,
  "pin": 4,
  "classType": "OneWireSensor",
  "deviceName": "Thermometer",
  "imageName": "thermo.jpg",
  "deviceColor": "70B829",
  "deviceState": false,
  "suspendTime": false,
  "deviceId": 1,
  "min": 10,
  "max": 50,
  "unit": ""
}
]
`>`

Will I receive the following log output:

`>`
Connected to	wifi
IP Address:	192.168.1.180
Loading	Devices
Found	devices: 1
Guru Meditation	Error: Core	 1 panic'ed	(LoadProhibited).	Exception	was	unhandled.
Core 1 register	dump:
PC			:	0x4000127a	PS			:	0x00060630	A0			:	0x800d82fa	A1			:	0x3ffb1ed0	
A2			:	0x3f401d46	A3			:	0x00000000	A4			:	0x0000000a	A5			:	0x0000005d	
A6			:	0x00000000	A7			:	0x3ffc2b80	A8			:	0x00000053	A9			:	0x00000000	
A10			:	0x00000000	A11			:	0x3f401e29	A12			:	0x0000000a	A13			:	0x00000001	
A14			:	0x3f401e92	A15			:	0xff000000	SAR			:	0x00000010	EXCCAUSE:	0x0000001c	
EXCVADDR:	0x00000000	LBEG		:	0x400014fd	LEND		:	0x4000150d	LCOUNT	:	0xffffffff	
Backtrace: 0x4000127a:0x3ffb1ed0 0x400d82f7:0x3ffb1ee0 0x400d85a7:0x3ffb1f40 0x400e1fbf:0x3ffb1fb0 0x4008930d:0x3ffb1fd0
Rebooting
`>`

Am I doing something wrong or is there an error in the code?
Can you help me.

mistergreen, 2020-06-04:
hey, try posting your problem over here. I'm not sure what the issue is.
https://github.com/Avilla83/WAAC

urkrossi, 2020-06-04:
> Hi urkrossi,
>
> I need to anticipate that the devices store and load is still in debug so I do not guarantee that it works 100%. Anyway the devices.json file is supposed to be written by the app and not manually. Could you try to delete it and let the firmware handle it?
>
> I am in the process of implementing the hardware so soon I will probably debug it.

Hi Avilla83,

 

I have now reinstalled my two boards. Both are correctly displayed on the home site after setting up a thermometer and saving. If I then restart the board, a device is loaded in the log with the correct settings that I have set. However, the thermometer is no longer displayed on the home site.

 

 

Log with configuration and storage of the device:

 

>> 

......ets Jun  8 2016 00:22:57

 

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0018,len:4

load:0x3fff001c,len:1044

load:0x40078000,len:8896

load:0x40080400,len:5816

entry 0x400806ac

Initializing SPIFFS...

SUCCESS - Found index.htm file.

Print config file...

{

  "ssid": "my ssid",

  "password": "my pass",

  "ip": [192, 168, 1, 180],

  "gateway": [192, 168, 1, 1],

  "subnet": [255, 255, 255, 0],

  "timeZone": 2,

  "ntpServer": "time.google.com"

}

Loading configuration...

{

  "ssid": "my ssid",

  "password": "my pass",

  "ip": [192, 168, 1, 180],

  "gateway": [192, 168, 1, 1],

  "subnet": [255, 255, 255, 0],

  "timeZone": 2,

  "ntpServer": "time.google.com"

}

Connecting to my ssid

.Connected to wifi

IP Address: 192.168.1.180

Loading Devices

Failed to read file, using default configuration

 

waiting for sync

Transmit NTP Request

Receive NTP Response

memory 253468

user pass

14177

set: and

Saving Devices

Serializing OneWireSensor

[{"numDevices":1},{"pin":4,"classType":"OneWireSensor","deviceName":"Thermometer","imageName":"thermo.jpg","deviceColor":"70B829","deviceState":false,"suspendTime":false,"deviceId":1,"min":0,"max":50,"unit":""}]

success ajax

 

<< 

 

Log after restarting the board:

 

>> 

ets Jun  8 2016 00:22:57

 

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0018,len:4

load:0x3fff001c,len:1044

load:0x40078000,len:8896

load:0x40080400,len:5816

entry 0x400806ac

Initializing SPIFFS...

SUCCESS - Found index.htm file.

Print config file...

{

  "ssid": "my ssid",

  "password": "my pass",

  "ip": [192, 168, 1, 180],

  "gateway": [192, 168, 1, 1],

  "subnet": [255, 255, 255, 0],

  "timeZone": 2,

  "ntpServer": "time.google.com"

}

Loading configuration...

{

  "ssid": "my ssid",

  "password": "my pass",

  "ip": [192, 168, 1, 180],

  "gateway": [192, 168, 1, 1],

  "subnet": [255, 255, 255, 0],

  "timeZone": 2,

  "ntpServer": "time.google.com"

}

Connecting to my ssid

....Connected to wifi

IP Address: 192.168.1.180

Loading Devices

Found devices: 1

[{"numDevices":1},{"pin":4,"classType":"OneWireSensor","deviceName":"Thermometer","imageName":"thermo.jpg","deviceColor":"70B829","deviceState":false,"suspendTime":false,"deviceId":1,"min":0,"max":50,"unit":""}]

waiting for sync

Transmit NTP Request

Receive NTP Response

memory 253324

user pass

46892

<< 

 

I hope we can find a way that I can use the controller.

It is a great project.

 

Kind regards

Avilla83, 2020-06-13:
Hi urkrossi,

I checked in the code and I didn't remember a simple fact: only the servo motor and the relay are completely stored and loaded! 
All the sensors are composed of a generic sensor class and a specific sensor class.
All the devices (except relays and servo motor) are stored as generic sensors and this is not enough to allocate them. So what is missing are the serializationa and deserialization methods in the specific one wire sensor class.

Said so there are some bugs in the relay too so now I am focusing on debugging the restore process with a simple relay. If I find some time I may add the one wire sensor too.

It would be good if you have a chance to test the code too, so we have a double check on my changes. I will let you know as soon as I have something working.

Regards

Avilla83, 2020-06-22:
Hi urkrossi,
I just made a commit in my branch configuration_storage, here the content:

Added One Wire store and load - partially tested
Fix for relay event handling not being properly triggered
Fix for configuration store not saving ip related info when configuration saved
Fix for servo motor pin not being properly set on servo library
Devices store fix, document was not cleared and so all devices were added to the old ones
 configuration_storage

The OneWireSensor class should be correctly saved and loaded but unfortunately I don't have a sensor of this type for a complete testing. If you can test please report the results.

I will test the events handling changes for Relay since they were not working in the previous code.

Regards

urkrossi, 2020-06-26:
Hello Avilla83,

this sounds good, I can test here with the help of DS18b20 One Wire sensors whether the configurations are saved and loaded again after restart.
Can they provide me with the customized files for testing?

Greeting

Avilla83, 2020-07-01:
Hi ukrossi,

I am not sure what you mean for customized files, do you mean the devices configuration?
If so I am using the ESP internal memory so it is not easy for me to extract files from there. 
What you can do is to add a one wire sensor, set the pin and see the result. All the other configurations are needed in case you want to control another device with the temperature value.

In the meanwhile my DS18B20 sensor arrived so I have been able to test it and it seems to work correctly.

I suggest you also to download from my stream since I uploaded some fixes there about devices persistent storage (incase you are using it).

Regards