mistergreen/WAAC #20: Need compilation verison

liwajaya, 2021-07-08:
hello everyone, is there anyone here who can help me, I need a BIN file and a spiff BIN, for Esp32-Devkit. I tried to compile with Ide 1.8 too many errors..

thank you

mistergreen, 2021-07-09:
What kind of errors are you getting when you compile?

liwajaya, 2021-07-09:
@mistergreen thanks for quick replay

compile "Esp32 Dev Module"
Arduino IDE 1.8.10

`sketch\WAAC_ESP32.ino.cpp.o:(.literal._ZZ16createHallSensorvENUlvE_4_FUNEv+0x0): undefined reference to `deviceDelegate'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._ZZ16createHallSensorvENUlvE_4_FUNEv+0x4): undefined reference to `DeviceDelegate::currentDevice()'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._ZZ16createHallSensorvENUlvE_4_FUNEv+0x8): undefined reference to `HallSensor::rpm()'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z10convertPinPc+0x34): undefined reference to `WebParser::contains(char*, char const*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z8loggedInv+0x10): undefined reference to `WebParser::clearBuffer(char*, int)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z8loggedInv+0x14): undefined reference to `WebParser::parseQuery(char*, char const*, char*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z9findColorPcS_+0x4): undefined reference to `WebParser::compare(char const*, char const*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z5setupv+0x48): undefined reference to `WWWsettings::begin()'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z5setupv+0x50): undefined reference to `WWWsettings::syncNTP()'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z11set8AdaPWMs10WiFiClient+0x24): undefined reference to `DeviceDelegate::findDevice(int)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z11set8AdaPWMs10WiFiClient+0x2c): undefined reference to `AdaFruitPWM8::setPWMs(int, int, int, int, int, int, int, int)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z22adaFruit8PWMAjaxOutput10WiFiClientP6Device+0x70): undefined reference to `Device::getPin()'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z22adaFruit8PWMAjaxOutput10WiFiClientP6Device+0x74): undefined reference to `AdaFruitPWM8::getI2C(int*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z22adaFruit8PWMAjaxOutput10WiFiClientP6Device+0x78): undefined reference to `AdaFruitPWM8::getPins(int*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z16saveAdaFruit8PWMP6Device+0x34): undefined reference to `AdaFruitPWM8::setI2C(int, int)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z16saveAdaFruit8PWMP6Device+0x38): undefined reference to `AdaFruitPWM8::setPins(int, int, int, int, int, int, int, int)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z18createAdaFruit8PWMv+0x10): undefined reference to `AdaFruitPWM8::AdaFruitPWM8(char*, int, int, int)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z18createAdaFruit8PWMv+0x14): undefined reference to `DeviceDelegate::addDevice(Device*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z18createAdaFruit8PWMv+0x18): undefined reference to `Device::setImageName(char*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z18createAdaFruit8PWMv+0x1c): undefined reference to `Device::setDeviceColor(char*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z15alertAjaxOutput10WiFiClientP6Device+0x18): undefined reference to `Alert::getEmail()'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z15alertAjaxOutput10WiFiClientP6Device+0x1c): undefined reference to `Alert::getSubject()'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z15alertAjaxOutput10WiFiClientP6Device+0x20): undefined reference to `Alert::getMessage()'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z9saveAlertP6Device+0xc): undefined reference to `WWWsettings::getToEmail()'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z9saveAlertP6Device+0x10): undefined reference to `Alert::setEmail(char*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z9saveAlertP6Device+0x14): undefined reference to `Alert::setSubject(char*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z9saveAlertP6Device+0x18): undefined reference to `Alert::setMessage(char*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z9saveAlertP6Device+0x1c): undefined reference to `Device::setDeviceState(bool)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z11createAlertv+0x4): undefined reference to `Alert::Alert(char*, int)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z15relayAjaxOutput10WiFiClientP6Device+0x8): undefined reference to `Relay::getInvert()'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z18inputMCPajaxOutput10WiFiClientP6Device+0x0): undefined reference to `InputMCP::getI2C(int*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z18relayMCPajaxOutput10WiFiClientP6Device+0x0): undefined reference to `RelayMCP::getI2C(int*)'

sketch\WAAC_ESP32.ino.cpp.o:(.literal._Z18relayPCAajaxOutput10WiFiClientP6Device+0x0): undefined reference to `RelayPCA::getI2C(int*)'
[... 1583 lines cut ...]
C:\Users\liwajaya1\Documents\WAAC_ESP32/WAAC_ESP32.ino:446: undefined reference to `WebParser::clearBuffer(char*, int)'

C:\Users\liwajaya1\Documents\WAAC_ESP32/WAAC_ESP32.ino:448: undefined reference to `WebParser::contains(char*, char const*)'

C:\Users\liwajaya1\Documents\WAAC_ESP32/WAAC_ESP32.ino:438: undefined reference to `WebParser::clearBuffer(char*, int)'

C:\Users\liwajaya1\Documents\WAAC_ESP32/WAAC_ESP32.ino:439: undefined reference to `WebParser::fileUrl(char*, char*)'

sketch\WAAC_ESP32.ino.cpp.o: In function `loop()':

C:\Users\liwajaya1\Documents\WAAC_ESP32/WAAC_ESP32.ino:241: undefined reference to `DeviceDelegate::loopDevices()'

C:\Users\liwajaya1\Documents\WAAC_ESP32/WAAC_ESP32.ino:244: undefined reference to `WWWsettings::check()'

C:\Users\liwajaya1\Documents\WAAC_ESP32/WAAC_ESP32.ino:262: undefined reference to `WebParser::clearBuffer(char*, int)'

sketch\WAAC_ESP32.ino.cpp.o: In function `Print::Print()':

C:\Users\liwajaya1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:48: undefined reference to `WebParser::WebParser()'

sketch\WAAC_ESP32.ino.cpp.o: In function `_GLOBAL__sub_I_dayName':

C:\Users\liwajaya1\Documents\WAAC_ESP32/WAAC_ESP32.ino:117: undefined reference to `WWWsettings::WWWsettings()'

collect2.exe: error: ld returned 1 exit status`

mistergreen, 2021-07-09:
Make sure to point the sketch location away from your typical arduino folder and to a new folder with the hardware/espressif compiler. I create the library folder in this as well. Make sure to place the library files in the new folder.